HarlemSquirrel/gnome-shell-extension-sensory-perception #5: Latest update killed functionality to show sensor data

George1896, 2018-10-31:
sensory perception worked fine until installing the latest update two day ago. It still shows SSD temp but no more CPU temp or fan rpm.

lm-sensors is installed and "sensors" output is like this:

~$ sensors
atk0110-acpi-0
Adapter: ACPI interface
Vcore Voltage:      +1.04 V  (min =  +0.85 V, max =  +1.60 V)
 +3.3 Voltage:      +3.38 V  (min =  +2.97 V, max =  +3.63 V)
 +5 Voltage:        +5.12 V  (min =  +4.50 V, max =  +5.50 V)
 +12 Voltage:      +12.16 V  (min = +10.20 V, max = +13.80 V)
CPU FAN Speed:     1285 RPM  (min =  600 RPM, max = 7200 RPM)
CHASSIS FAN Speed:    0 RPM  (min =  600 RPM, max = 7200 RPM)
POWER FAN Speed:      0 RPM  (min =  600 RPM, max = 7200 RPM)
CPU Temperature:    +37.0°C  (high = +60.0°C, crit = +95.0°C)
MB Temperature:     +36.0°C  (high = +45.0°C, crit = +95.0°C)

HarlemSquirrel, 2018-11-02:
Thank you for reporting this @George1896 ! I think 4d23e7a53fe8c8e5cc5d318fb09b928ae44508e6 should fix this. Are you able to test this?

nathansamson, 2018-11-08:
I have the same problem, updated (which should include the fix from master), but still an issue

nathansamson, 2018-11-08:
```
sensors
ath10k_hwmon-pci-0200
Adapter: PCI adapter
temp1:        +41.0°C  

dell_smm-virtual-0
Adapter: Virtual device
Processor Fan: 2517 RPM
Video Fan:     2498 RPM
CPU:            +59.0°C  
Ambient:        +52.0°C  
Ambient:        +48.0°C  
Other:          +37.0°C  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +25.0°C  (crit = +107.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +52.0°C  (high = +100.0°C, crit = +100.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +56.5°C  
```

becquerel, 2018-11-08:
it fixed the problem for me

@nathansamson maybe try reloading gnome ? It did the trick for me, ALT + F2, type `r` and [ENTER]

George1896, 2018-11-09:
@HarlemSquirrel

Sorry for my late reply.

I removed the extension, did a git clone yesterday and tried your changes. They work fine for me now.

Thank you for fixing it :)

nathansamson, 2018-11-09:
@becquerel Today it works, i assume restarting gnome shell as you suggested might have helped as well.
Bit weird that a reinstall (which removed the error from my bar even for a few seconds) did not have the same effect

HarlemSquirrel, 2018-11-09:
Thank you for confirming the fix! v7 includes the fix and should now be available through the [GNOME extensions page](https://extensions.gnome.org/extension/1145/sensory-perception/).

GNOME does not seem to do a good job of reloading extensions when they change. Running on Xorg instead of Wayland enables restarting the shell with [Alt] + [F2] and entering "r".

closed 2018-11-09 by HarlemSquirrel