HarlemSquirrel/gnome-shell-extension-sensory-perception #6: Latest updates show error and request running as root

j00bar, 2018-11-10:
Using F28, this extension worked fine until recently. Now it reads "Error" and on click recommends running as root.

HarlemSquirrel, 2018-11-11:
Hi, could you share any relevant log output? Something like `journalctl --since="2018-11-10 15:00" | grep sensory-perception` should work. 

Also, please make sure you are running the latest version v7 by visiting the [GNOME extensions page](https://extensions.gnome.org/extension/1145/sensory-perception/) and looking for an update button.

jjakob, 2018-12-27:
I had the same problem, upgrading the extension via the extensions page worked. Gnome should add an auto-update (or update check) feature. Thanks.

HarlemSquirrel, 2019-03-02:
I'm going to close this since it should be working now with the latest version.

closed 2019-03-02 by HarlemSquirrel