HarlemSquirrel/gnome-shell-extension-sensory-perception #7: Setting are not working

thefeiter, 2018-12-24:
On Ubuntu 18.04.01 LTS I can't open the sensor settings. Just a window with this content pops up:

GLib.ShellError: Text war leer (oder enthielt nur Leerraum)

Stack trace:
  SensorsPrefsWidget<._getSensorsLabels@/home/marvin/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:164:29
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  SensorsPrefsWidget<._init@/home/marvin/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:113:5
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  buildPrefsWidget@/home/marvin/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:245:18
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:91:22
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:243:17
  wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
  main@resource:///org/gnome/shell/extensionPrefs/main.js:397:5
  @<main>:1:43

HarlemSquirrel, 2018-12-28:
Hi @thefeiter Which version of this extension do you have installed?

Please check for any available updates at https://extensions.gnome.org/extension/1145/sensory-perception/

HarlemSquirrel, 2019-03-02:
Let me know if this is still a problem for you on the latest version 7.

closed 2019-03-02 by HarlemSquirrel

saildata, 2019-03-16:
First, I'd like to say CONGRATS :tada:  on being the first working GS Extension since the massive outage yesterday when 3.32 was released to main in Arch Linux. I hope that others are able to use this as an example of how to un-break their extensions.

So ... the settings are still an issue on v7 installed locally (50f3b6f8b310babf2be475d4af88d43f66ba0505).

The only logged item WHEN I click on preferences from `gnome-tweaks` is
```
Mar 16 13:17:20 pc01 gnome-shell-ext[6202]: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
```
which doesn't seem like it would be a *show stopper*, but I thought I'd include it just in case.

Here is what it looks like on my system when I open settings:

[image]