HarlemSquirrel/gnome-shell-extension-sensory-perception #10: Icons/Switches are not properly scaled

zzzardoz, 2019-04-16:
**Describe the bug**
Since Gnome 3.32 icons and switches are way to big. (Please see screenshots.) I'm on latest Arch and I have a 4K screen with 200% scale. 

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux

HarlemSquirrel, 2019-04-16:
Hi @zzzardoz , what version of the extension are you using? Can you try the master branch? 
Thanks!

zzzardoz, 2019-04-16:
@HarlemSquirrel: Just tried master branch, but it makes no difference. I was using the latest version from Gnome Extensions website.

Thanks for the fast response! :-)

HarlemSquirrel, 2019-04-16:
Does changing scaling to 100% make a difference?

zzzardoz, 2019-04-17:
Unfortunately not, still the same...

ellperro, 2019-04-23:
same here, ubuntu 19.04 gnome 3.32.0 monitor 1920x1200 (dual monitor setup) no scaling
Big icons -> doesn't fit the screen

Screenshot:
[image]

HarlemSquirrel, 2019-04-26:
I think a0acd43540167ebe2dace2db984ed7670aa336b9 should fix this. Could someone please try the latest code from master and let me know?

zzzardoz, 2019-04-26:
@HarlemSquirrel: That indeed fixes the icons. But I'd suggest for going with 16px. Switches in prefs are still too large.

HarlemSquirrel, 2019-05-04:
I've released a new version which includes the menu icon size reduction. It looks like the stretched switches in the preferences menu is related to how the grid is used. Haven't found a way to fix it yet.

HarlemSquirrel, 2021-01-03:
This should be fixed

closed 2021-01-03 by HarlemSquirrel