HarlemSquirrel/gnome-shell-extension-sensory-perception #13: It causes sutter video playing in windowed mode.

laichiaheng, 2019-05-16:
**Describe the bug**
If I watch videos in windowed mode, no matter in Firefox or MPV player, the video stutters a lot.

**To Reproduce**
Steps to reproduce the behavior:
1. enable this extension
2. Play videos in windowed mode.
4. See error

**Expected behavior**
Don't affect the video playing.

**Desktop (please complete the following information):**
 - OS: [Manjaro]

HarlemSquirrel, 2019-05-30:
I think this may be related to running the `sensors` command. On my Ryzen desktop the execution hangs for about half a second. 

```
➤  time sensors
amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.93 V
Fury X:       938 RPM  (min =  500 RPM, max = 3300 RPM)
Fury X:       +28.0°C  (crit = +79.0°C, hyst = -273.1°C)
power1:       17.17 W  (cap = 300.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
CPU:          +30.9°C  (high = +70.0°C)
Tctl:         +50.9°C

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

sensors  0.00s user 0.50s system 99% cpu 0.505 total
```

I'm thinking that instead of using `sensors`, this extension could read values from sysfs. 

```
➤  time egrep ".*" `ls /sys/devices/**/hwmon/**/*_{input,label}`
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/fan1_input:938
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/freq1_input:560530000
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/freq1_label:sclk
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/freq2_input:500000000
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/freq2_label:mclk
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/in0_input:925
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/in0_label:vddgfx
/sys/devices/pci0000:00/0000:00:03.1/0000:0b:00.0/hwmon/hwmon2/temp1_input:28000
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon0/temp1_input:40250
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon0/temp1_label:Tdie
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon0/temp2_input:60250
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon0/temp2_label:Tctl
/sys/devices/platform/eeepc-wmi/hwmon/hwmon1/fan1_input:0
/sys/devices/platform/eeepc-wmi/hwmon/hwmon1/fan1_label:cpu_fan
egrep ".*" `ls /sys/devices/**/hwmon/**/*_{input,label}`  0.00s user 0.00s system 96% cpu 0.005 total
```

HarlemSquirrel, 2019-06-14:
I was able to get sensors to return immediately by ignoring the power sensor.

`/etc/sensors.d/amdgpu-pci-0b00`
```
chip "amdgpu-pci-0b00"
  label temp1 "Fury X"
  label fan1 "Fury X"
  ignore power1
```

```
➤  time sensors
amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.93 V
Fury X:       937 RPM  (min =  500 RPM, max = 3300 RPM)
Fury X:       +29.0°C  (crit = +79.0°C, hyst = -273.1°C)

k10temp-pci-00c3
Adapter: PCI adapter
CPU:          +45.2°C  (high = +70.0°C)
Tctl:         +65.2°C

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

sensors  0.00s user 0.00s system 95% cpu 0.005 total
```

closed 2021-11-28 by HarlemSquirrel