HarlemSquirrel/gnome-shell-extension-sensory-perception #14: Settings are not working in GNOME Shell 3.32.1

rovellipaolo, 2019-06-14:
**Describe the bug**
When opening the settings, an error is displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://extensions.gnome.org/local/
2. Click on 'Configure this extension' button
3. An error is displayed

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.04
 - GNOME Shell: 3.32.1

**Additional context**
GLib.ShellError: Text was empty (or contained only whitespace)

Stack trace:
  _getSensorsLabels@/home/provelli/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:195:29
  _init@/home/provelli/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:143:5
  buildPrefsWidget@/home/provelli/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:276:18
  _selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:75:22
  _onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:320:17
  main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
  @<main>:1:43

atanaschristov, 2019-06-19:
I'm having the same issue:

[image]

HarlemSquirrel, 2019-06-22:
What version of the extension are you running? Have you tried manually deleting the extension and then installing again?

To manually delete the extension you can run the command below, log out, and then log back in. 

```
rm -r ~/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io
```

rovellipaolo, 2019-06-26:
I was running version _10_ (the latest one available here: https://extensions.gnome.org/extension/1145/sensory-perception/).

But, yes, manually deleting the extension actually fixed the issue.

closed 2019-06-26 by rovellipaolo