HarlemSquirrel/gnome-shell-extension-sensory-perception #15: can't disable wrong items values

eliaspizarro, 2019-06-14:
I've a wrong lecture from my SSD, should be an option to disable some measures
[image]

HarlemSquirrel, 2019-06-22:
This uses the information returned from the `sensors` command. We can hide individual sensor output similar to what I describe in https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/issues/13#issuecomment-501935204

There is a full [sensor.conf manual](https://linux.die.net/man/5/sensors.conf) for doing this. You may also have a local copy accessible with `man sensors.conf`

GaVenga, 2020-07-28:
Problem similar (/etc/sensors.d/nct6798-isa-0290):
chip "nct6798-isa-0290"
ignore AUXTIN0
ignore AUXTIN1
ignore AUXTIN2
ignore AUXTIN3
label AUXTIN0 "Netzteil"
label fan1 "Gehäuse"

This does NOT ignore "AUXTIN0" to "AUXTIN3"
but labels fan1 correctly