HarlemSquirrel/gnome-shell-extension-sensory-perception #16: Something's gone wrong (Not working on Ubuntu 19.04)

cassv24, 2019-07-08:
**Describe the bug**
When I click the settings in the menu I get the following error:

`The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: Text was empty (or contained only whitespace)
```

Stack trace:
```
_getSensorsLabels@~/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:195:29
_init@~/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:143:5
buildPrefsWidget@~/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:276:18
_selectExtension@resource:///org/gnome/shell/extensionPrefs/main.js:75:22
_onCommandLine@resource:///org/gnome/shell/extensionPrefs/main.js:320:17
main@resource:///org/gnome/shell/extensionPrefs/main.js:635:5
@<main>:1:43
```
`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the menu
2. Click on settings
3. See error

**Expected behavior**
To open the settings panel

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 19.04

HarlemSquirrel, 2019-07-11:
Hi @cassv24 , could you share the output of `sensors`?

cassv24, 2019-07-14:
I'm not sure what are you asking for, but this is the output of the program:

[image]

HarlemSquirrel, 2019-07-24:
@cassv24 I'm looking for the output of running the command `sensors` in a terminal. 

It should look something like this:

```
➤  sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +56.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +56.0°C  (crit = +120.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +29.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +44.5°C
```

cassv24, 2019-07-28:
Hi, sorry for the delay.

after installing `lm-sensors` it's working now. 

You should add _lm-sensors_ as an explicit dependency in your README.md as part of the installation instructions.

closed 2019-07-28 by cassv24

HarlemSquirrel, 2019-07-28:
I'm glad that worked! I just updated the README. Thanks for the suggestion :)