HarlemSquirrel/gnome-shell-extension-sensory-perception #17: Translation in UI not shown

Etamuk, 2019-07-25:
**Describe the bug**
I am using my computer with german interface. After installing this extension the UI is still in english. The translation file de.po is availible in the extension folder.

**To Reproduce**
Steps to reproduce the behavior:
1. You need a system language which is not english, f.e. german
2. Install this ext.
3. After relogin or Alt+F2 + r the UI of this ext. is still english

**Expected behavior**
See this ext. with german interface
**Screenshots**
---

**Desktop (please complete the following information):**
 - Ubuntu 19.04
 - Gnome 3.32.1

jose1711, 2019-09-23:
@Etamuk please test with https://github.com/jose1711/gnome-shell-extension-sensory-perception/commit/62c59897b6d6035fbefeb2fd794dc82cb3ed6674

HarlemSquirrel, 2021-11-28:
It's this still broken in latest v16?

jose1711, 2021-11-28:
Tested this on a fresh installed via extensions.gnome.org. Same problem, all is in English. Placing the `.mo` and `.po` files into `locale/LANG/sk/LC_MESSAGES` and minor changes in `prefs.js` as done in test branch still work.

Etamuk, 2021-11-28:
I installed the latest V16 from https://extensions.gnome.org. Still the whole UI in English.

HarlemSquirrel, 2021-11-28:
@jose1711 could you open a pull request with those changes?

jose1711, 2021-11-28:
PR raised. Feel free to comment and/or edit directly.

PR #45 (2021-11-29): Fix broken l10n
Fixes #17
commit 421b82f4 (2021-11-29): Fix broken l10n

Fixes #17
closed 2021-11-29 by HarlemSquirrel