HarlemSquirrel/gnome-shell-extension-sensory-perception #26: Sensory Perception extension reports the wrong temperature parameter for some SSDs

paulgir, 2020-04-14:
Recently I rebuilt my computer with a new motherboard and a M.2 format system drive.
Sensory Perception was reporting the drive temperature as 100 ℃ 
but in CLI hddtemp was showing 33 ℃ and measuring the actual drive with an IR thermometer agrees with 33 ℃.
I have found out why this is. 
Looking at S.M.A.R.T. data,there are two temperature attributes: 
Attribute ID 194 Temperature which shows 33 ℃ 
and Attribute ID 231 which is Temperature for HDDs or Life Left for SSDs      100  -  equates to a new drive.
So ,in my case Sensory Perception is reporting the Life Left value as Temperature.
This page shows the S.M.A.R.T. data information:https://en.wikipedia.org/wiki/S.M.A.R.T.#Known_ATA_S.M.A.R.T._attributes

The SSD is a KINGSTON SA400M8120G
Motherboard is: Asus Prime B-450M-A
CPU: AMD® Ryzen 5 2600 six-core processor × 12
Desktop: Ubuntu 19.10
[image]

HarlemSquirrel, 2020-04-14:
I see two temperature readings of 29 and 99 in the screenshot. The summary section of the disk utility shows the much higher 99 degrees and must be what this extension is reading.

Could you also share the terminal output of the `hddtemp` command?

paulgir, 2020-04-14:
~$ sudo hddtemp /dev/sdc
/dev/sdc: KINGSTON SA400M8120G: 26°C

HarlemSquirrel, 2020-04-14:
Interesting. It's good to see that `hddtemp` reports the correct temperature but that requires `sudo`. We'll have to see if there's a better way to get this data.

HarlemSquirrel, 2020-04-15:
So when `hddtemp` cannot but run as the current user, we try to get the temperature from the `hddtemp` service which defaults to port 7634 and can be seen with netcat like

```
➤  nc localhost 7634
|/dev/sda|ADATA SU800NS38|40|C|
```

@paulgir do you get something like this?

paulgir, 2020-04-15:
I'm not familiar with netcat.
If I run nc localhost 7634 I get no output

paulgir, 2020-04-15:
Ok, I did a little googling.
I ran:       `ps -eo 'tty,pid,comm' | grep ^?`       to see if hddtemp is running as a daemon and it appears it isn't.
I ran:       `hddtemp -d /dev/sda`     then:       `nc localhost 7634`     and I get      `|/dev/sda|???|ERR|*|`

HarlemSquirrel, 2020-04-15:
How about if you start the daemon with `sudo systemctl start hddtemp`

paulgir, 2020-04-15:
No  output from `nc localhost 7634` if I  start the daemon with `sudo systemctl start hddtemp`

wildart, 2020-10-08:
Have the same problem, `Patriot Burst` SSD reports 100C, but according to `hddtemp` service, it's 33C.

[image]

`nc localhost 7634` returns 
```
|/dev/sda|WDC WD60EZRZ-00GZ5B1|29|C||/dev/sdb|ST3250310AS|33|C||/dev/sdc|Patriot Burst|33|C|
```

Interestingly, that `smartctl` reports attribute `194 - Temperature_Celsius` correctly `33`.
```
$ sudo smartctl --attributes /dev/sdc                                                                                                                
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.0-48-generic] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000b   100   100   050    Pre-fail  Always       -       0
  9 Power_On_Hours          0x0012   100   100   000    Old_age   Always       -       2059
 12 Power_Cycle_Count       0x0012   100   100   000    Old_age   Always       -       247
168 SATA_Phy_Error_Count    0x0012   100   100   000    Old_age   Always       -       0
170 Bad_Blk_Ct_Erl/Lat      0x0003   100   100   010    Pre-fail  Always       -       0/62
173 MaxAvgErase_Ct          0x0012   100   100   000    Old_age   Always       -       20 (Average 7)
192 Unsafe_Shutdown_Count   0x0012   100   100   000    Old_age   Always       -       14
194 Temperature_Celsius     0x0023   067   067   000    Pre-fail  Always       -       33 (Min/Max 33/33)
218 CRC_Error_Count         0x000b   100   100   050    Pre-fail  Always       -       0
231 SSD_Life_Left           0x0013   100   100   000    Pre-fail  Always       -       99
241 Lifetime_Writes_GiB     0x0012   100   100   000    Old_age   Always       -       1128
```
However, another attribute `231 - SSD_Life_Left `, in `gnome-disks`, is reported as `Temperature` parameter.

[image]

Running Ubuntu 20.04 with GNOME 3.36.

paulgir, 2020-10-08:
> Have the same problem, `Patriot Burst` SSD reports 100C, but according to `hddtemp` service, it's 33C.
>
> [image]
>
> `nc localhost 7634` returns
>
> ```
> |/dev/sda|WDC [...]

Is this an M.2 format SSD?
I built a computer for a friend with the same mobo but a discrete SSD  and that one displayed the temp correctly in the extension.

wildart, 2020-10-08:
It's SATA disk.

antoniovazquezblanco, 2024-05-05:
This has been fixed in an independent fork of the original lib that it is not currently recognized as official upstream by maintainers at https://github.com/libatasmart/libatasmart.