HarlemSquirrel/gnome-shell-extension-sensory-perception #27: Settings gui not working

frakbe, 2020-06-25:
**Describe the bug**
Settings gui not working

**To Reproduce**
Steps to reproduce the behavior:
1. Click settings
2. Error displayed

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

**Additional context**
trace:
The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: Il testo era vuoto (oppure conteneva unicamente spazi bianchi)
```

Stack trace:
```
_getSensorsLabels@/home/franco/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
_init@/home/franco/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
buildPrefsWidget@/home/franco/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

HarlemSquirrel, 2020-06-25:
What version of this extension are you running?

frakbe, 2020-06-26:
Hi Kevin,
sorry, my fault.
I've installed your extension (v12) from gnome-shell extensions website, using firefox addon. This way, the extension was installed without checking of required software.
Reading the Readme.md file I dicovered it wants lm-sensor in order to work.
Installing lm-sensor resolve the 'issue'.
Suggestion for possible ui ehnancement; try&catch if lm-sensors binaries is installed on system when clicking on "settings" may help final users.

Thanks for all
Frakbe

HarlemSquirrel, 2020-06-27:
No problem. I'm glad you worked it out! Good suggestion. I well a I'll request for that! 🙂

closed 2020-06-27 by HarlemSquirrel