HarlemSquirrel/gnome-shell-extension-sensory-perception #30: No more display with gnome-shell 3.38

grandtoubab, 2020-10-07:
No more work in gnome shell 3.38 / Debian 11 Bullseye

No display of temperature

logs are filled with this messages

journalctl -f | grep sensory-perception
oct. 07 22:04:51 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:05:06 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:05:21 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:05:36 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:05:51 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:06:06 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:06:21 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:06:36 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:06:51 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:07:06 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:07:21 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:07:36 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:07:51 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:08:06 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:08:21 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:08:36 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null
oct. 07 22:08:51 debian gnome-shell[7362]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _readStdout: TypeError: meta is null

[image]

HarlemSquirrel, 2020-10-08:
Hi, is this working now for you with version 13?

grandtoubab, 2020-10-08:
Hello, 
yes updated on 08 october fixed the issue
[image]

closed 2020-10-08 by grandtoubab