HarlemSquirrel/gnome-shell-extension-sensory-perception #31: Unable to install in Ubuntu 20.04

abickerton-upc, 2020-10-08:
**Describe the bug**
Extension is not usable in Ubuntu 20.04.
Fails to update in https://extensions.gnome.org/extension/1145/sensory-perception/

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://extensions.gnome.org/extension/1145/sensory-perception/
2. Click on "On slider "
3. Select install on dialog
4. Error shown

**Expected behavior**
Extension is installed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

$ gnome-shell --version
GNOME Shell 3.36.4

HarlemSquirrel, 2020-10-10:
Could you please run the following command in a terminal, try again to install the extension, and share the output here?

```
journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f | grep sensory-perception
```

abickerton-upc, 2020-10-12:
The above command shows no output. Either something updated in the background or there was an update :-) somewhere.
Anyways, the extension appears to be working again. 

Thank you for your efforts.

closed 2020-10-12 by abickerton-upc