HarlemSquirrel/gnome-shell-extension-sensory-perception #32: setting can't be display

ddyllen-z, 2020-10-11:
**Describe the bug**
click settings display an error:can't display setting.

**To Reproduce**
click settings display an error:can't display setting.

**Expected behavior**
no

**Screenshots**
[image]

**Desktop:**
 - OS: ubuntu
 - Version : 20.04.1

**Additional context**

technical details:

The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: 文本为空(或仅含空白字符)
```

Stack trace:
```
_getSensorsLabels@/home/dyllen/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
_init@/home/dyllen/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
buildPrefsWidget@/home/dyllen/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

```
pc:~$ journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f | grep sensory-perception
10月 11 23:07:06 dyllen-MS-7C02 gnome-shell[2156]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _init: Error: Array element (type filename) may not be null
10月 11 23:07:21 dyllen-MS-7C02 gnome-shell[2156]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _init: Error: Array element (type filename) may not be null
10月 11 23:07:36 dyllen-MS-7C02 gnome-shell[2156]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _init: Error: Array element (type filename) may not be null
10月 11 23:07:51 dyllen-MS-7C02 gnome-shell[2156]: sensory-perception@HarlemSquirrel.github.io: ERROR Future _init: Error: Array element (type filename) may not be null

```

HarlemSquirrel, 2020-10-12:
Hi @ddyllen-z please share the output of `sensors` command.

ddyllen-z, 2020-10-13:
@HarlemSquirrel sorry, i don't know that `lm-sensors` must be installed.The temperature was normal before. It's normal now.tks.Thanks.

closed 2020-10-13 by ddyllen-z