HarlemSquirrel/gnome-shell-extension-sensory-perception #33: Gnome Tweak Option error

AFwcxx, 2020-10-29:
The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: Text was empty (or contained only whitespace)
```

Stack trace:
```
_getSensorsLabels@/home/afwcxx/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
_init@/home/afwcxx/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
buildPrefsWidget@/home/afwcxx/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

HarlemSquirrel, 2020-10-29:
Hi, thanks for the report. Could you please share the version of GNOME and lm-sensors you have installed?

AFwcxx, 2020-10-29:
> Hi, thanks for the report. Could you please share the version of GNOME and lm-sensors you have installed?

GNOME version: 3.36.3
But you've just gave me a clue. `lm-sensors` was not installed. Once installed, the issue is gone.
Thank you!

closed 2020-10-29 by AFwcxx