HarlemSquirrel/gnome-shell-extension-sensory-perception #34: Settings can't be opened

vadi2, 2021-01-02:
**Describe the bug**
```
The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: Text was empty (or contained only whitespace)
```

Stack trace:
```
_getSensorsLabels@/home/vadi/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
_init@/home/vadi/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
buildPrefsWidget@/home/vadi/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

```

**To Reproduce**
Install on Ubuntu 20.04 LTS and go to Settings

**Expected behavior**
Settings opens.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu]
 - Version [e.g. 18.04]

**Additional context**
Add any other context about the problem here.

HarlemSquirrel, 2021-01-03:
Hi, what version of lm-sensors do you have?

vadi2, 2021-01-03:
Latest in Ubuntu 20.04 LTS:

```
$ apt policy lm-sensors
lm-sensors:
  Installed: 1:3.6.0-2ubuntu1
  Candidate: 1:3.6.0-2ubuntu1
  Version table:
 *** 1:3.6.0-2ubuntu1 500
        500 http://nl.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
        100 /var/lib/dpkg/status
```

vadi2, 2021-01-03:
I didn't have lm-sensors setup, that was the issue. The description on the page didn't mention anything.

HarlemSquirrel, 2021-01-03:
I've updated the bug report template to include lm-sensors version. Hopefully that will be more helpful in the future.

closed 2021-01-03 by HarlemSquirrel

vadi2, 2021-01-03:
I think it would be a good idea to mention the lm-sensors requirement either in the description or when you install the applet. Discovering the dependency at bug report time isn't optional, by far not everyone will bother to submit a report...

HarlemSquirrel, 2021-01-03:
Just added that to the description 5e664174e7ad17bf5bcc0099ac94fd528a071f3d