HarlemSquirrel/gnome-shell-extension-sensory-perception #35: Not able to see settings for the extension

xames3, 2021-03-15:
I just installed the extension and though of changing couple of things, but the extension popped up with a failure message.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the extension settings from Panel shortcut.

Here is the log

The settings of extension sensory-perception@HarlemSquirrel.github.io had an error:
```
GLib.ShellError: Text was empty (or contained only whitespace)
```

Stack trace:
```
_getSensorsLabels@/home/xa/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
_init@/home/xa/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
buildPrefsWidget@/home/xa/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
_init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:13
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:175:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
run@resource:///org/gnome/gjs/modules/script/package.js:222:19
start@resource:///org/gnome/gjs/modules/script/package.js:206:5
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

HarlemSquirrel, 2021-03-22:
Could you please share the output of `sensors` command?

xames3, 2021-03-22:
The above is what I got when I clicked, let me know what else should I provide you in particular. If this seems to be an open/persistent issue then feel free to close it.

HarlemSquirrel, 2021-03-22:
To help debug this, could you please run `sensors` in a terminal and share the output here? Thank you!

xames3, 2021-03-24:
Yes, I got this after running the said command, 

```bash
$  sensors
ath10k_hwmon-pci-0200
Adapter: PCI adapter
temp1:        +38.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +50.0°C  (high = +100.0°C, crit = +100.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          16.61 V  
curr1:         0.00 A  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +51.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +44.9°C  (low  =  -5.2°C, high = +79.8°C)
                       (crit = +84.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

```

HarlemSquirrel, 2021-04-10:
What version of GNOME are you on?

xames3, 2021-04-10:
GNOME version is 3.36.8

HarlemSquirrel, 2021-04-13:
I'm a bit stumped on this one. Assuming you're on v13, the line in the backtrack is reading from `sensors` command and is reporting the output is blank/empty.

https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/blob/v13/prefs.js#L200

This usually happens if `sensors` is not installed. Could you try manually install v12 and see if you get the same issue?

HarlemSquirrel, 2021-06-06:
@xames3 could you run this from a terminal and share the output?

```
gjs --command="var sensorsProg = imports.gi.GLib.find_program_in_path('sensors'); log('typeof: ' + (typeof sensorsProg)); log(sensorsProg);"
```

HarlemSquirrel, 2021-07-05:
@xames3 @xzj8b3
Can you try running the code in the latest master and let me know if that works for you?

xzj8b3, 2021-07-09:
> @ xames3 @ xzj8b3
> Puoi provare a eseguire il codice nell'ultimo master e fammi sapere se funziona per te?

Sorry delay, now it seems to be working but not showing fan speeds or anything.

I'm on Gnome 40 with Fedora 34

HarlemSquirrel, 2021-07-09:
Can you share the output of `sensors` from your terminal?

timur-tabi, 2021-10-31:
I was getting this problem and then I installed lm-sensors and now it works.  It would be handy if the extension could detect that lm-sensors is not installed instead of just crashing.

HarlemSquirrel, 2021-11-01:
@timur-tabi That's what I attempted to do in the last code but it's not released yet. Can you confirm that the code from the latest master branch allows you to load the preferences when lm-sensors is not installed?

HarlemSquirrel, 2021-11-28:
There's a new release out for GNOME 40 and 41 that should better handle missing sensors program.

Tyrben, 2023-03-11:
Gnome 3.38
Debian 11

> GLib.ShellError: Le texte était vide (ou ne contenait que des espaces)
> Stack trace:
>   _getSensorsLabels@/home/---/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:200:40
>   _init@/home/---/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:149:14
  buildPrefsWidget@/home/---/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
>   _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:207:40
>   OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
>   asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:132:22
>   run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
>   main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
>   run@resource:///org/gnome/gjs/modules/script/package.js:222:19
>   start@resource:///org/gnome/gjs/modules/script/package.js:206:8
>   @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17

So I run your command:
> gjs --command="var sensorsProg = imports.gi.GLib.find_program_in_path('sensors'); log('typeof: ' + (typeof sensorsProg)); log(sensorsProg);"
> Gjs-Message: 20:15:46.262: JS LOG: typeof: object
> Gjs-Message: 20:15:46.262: JS LOG: null

Some on internet mention this error when switching of js version.

HarlemSquirrel, 2023-03-14:
You'll need to install lm-sensors