HarlemSquirrel/gnome-shell-extension-sensory-perception #36: Out of date for GNOME 40

rffontenelle, 2021-04-07:
**Describe the bug**
It is disable on GNOME 40 with state "out of date", also the Extensions viewer shows an error containing "TypeError: Gtk.RadioButton is not a constructor" (see full error log below)

**To Reproduce**
Steps to reproduce the behavior:
1. Update GNOME to release 40
2. Sensory Perception won't start
3. Run ```gnome-extensions info sensory-perception@HarlemSquirrel.github.io```
4. See "State: OUT OF DATE"
5. Open Extensions app
6. Browse the Sensory Perception extension
7. See error message

**Expected behavior**
To have Sensory Perception up and running.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux 64-bit
 - Version: rolling release
 - Version of lm-sensors (or lm_sensors): 3.6.0

**Additional context**

Error message displayed in Extensions app:

```
TypeError: Gtk.RadioButton is not a constructor

Stack trace:
  _init@/home/rafael/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:84:33
  buildPrefsWidget@/home/rafael/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io/prefs.js:281:20
  _init@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:209:40
  OpenExtensionPrefsAsync/<@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:122:28
  asyncCallback@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:115:22
  run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:177:20
  main@resource:///org/gnome/Shell/Extensions/js/main.js:19:13
  run@resource:///org/gnome/gjs/modules/script/package.js:206:19
  start@resource:///org/gnome/gjs/modules/script/package.js:190:8
  @/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
```

Output of `gnome-extensions info sensory-perception@HarlemSquirrel.github.io`:

```
sensory-perception@HarlemSquirrel.github.io
  Name: Sensory Perception
  Description: Shows CPU temperature, disk temperature, video card temperature, voltage and fan RPM (forked from xtranophilist/gnome-shell-extension-sensors)
  Path: /home/rafael/.local/share/gnome-shell/extensions/sensory-perception@HarlemSquirrel.github.io
  URL: https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception
  Version: 13
  State: OUT OF DATE
```

HarlemSquirrel, 2021-04-10:
New release v14 should fix this. It was just submitted for approval but you can download the release and manually install now.

HarlemSquirrel, 2021-04-12:
v14 has been approved so closing this.

closed 2021-04-12 by HarlemSquirrel

rffontenelle, 2021-04-12:
Installed, thanks!