HarlemSquirrel/gnome-shell-extension-sensory-perception #42: Displayed names are useless

timur-tabi, 2021-11-02:
I'm running Ubuntu 21.10, and the names displays on the extension are completely useless.  Here is the output of 'sensors':

```
nvme-pci-0800
Adapter: PCI adapter
Composite:    +33.9°C  (low  =  -5.2°C, high = +83.8°C)
                       (crit = +87.8°C)

amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:        1.34 V  
vddnb:       999.00 mV 
edge:         +35.0°C  
power1:        4.00 mW 

nvme-pci-0300
Adapter: PCI adapter
Composite:    +43.9°C  (low  =  -0.1°C, high = +74.8°C)
                       (crit = +79.8°C)
```
And here is what the drop-down shows:

[image]

So, a few things wrong here:

1) 'sensors' doesn't show the SSD drive, so where does that come from?
2) Using "Composite" for the NVME drives is dumb.  What does that even mean, and why would you pick that string instead of "nvme-pci-0800"?
3) The extension gets the two "Composite" entries confused.  Sometimes when I tell it to display the temp of one "Composite", it actually displays the temp of the other one.

What we need are two things:

1) The extension needs to use something other than the name to distinguish among different sensors, since the name could be duplicated.

2) It also needs to allow me to specify my own name for a given sensor, and not use whatever name 'sensors' spits out.

3) Frankly, the average of all temperatures is a pointless value to display.

HarlemSquirrel, 2021-11-10:
Thanks for taking the time to open this issue! 
I don't have much time to work in this myself but I welcome a pull request to improve the names and offer a way to specify custom names.

sojusnik, 2022-07-25:
[This](https://github.com/corecoding/Vitals) extension has better labels, maybe it can serve as an inspiration.

timur-tabi, 2022-07-25:
Thanks, Vitals is vastly superior, and even replaces some other extensions. I'll just use that instead.