HarlemSquirrel/gnome-shell-extension-sensory-perception #43: Gnome 41 support?

Netherquark, 2021-11-11:
This extension was working perfectly with Fedora 34/Gnome 40 but its not supported on Fedora 35/Gnome 41. Hope the dev can take the time to update it :)

nlincke, 2021-11-14:
Upvote! I would also love to see an update to solve this issue...

VergeDX, 2021-11-23:
`sed -i 's/"40"/"40", "41"/g' metadata.json` works for me fine : )

Netherquark, 2021-11-24:
> `sed -i 's/"40"/"40", "41"/g' metadata.json` works for me fine : )

This means its forward-compatible? thats great! So the dev would just need to publish a new release with this change right? @HarlemSquirrel

HarlemSquirrel, 2021-11-25:
Published [v15](https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/releases/tag/v15) and submitted for review

HarlemSquirrel, 2021-11-28:
Had to make some more changes from review on v15 but now v16 is live.

closed 2021-11-28 by HarlemSquirrel