HarlemSquirrel/gnome-shell-extension-sensory-perception #46: No values for fan speed and voltage

lemrm, 2021-12-22:
**Describe the bug**

in settings fan speed an voltage is enabled but section and values are not shown

**To Reproduce**

installed by chome-web-extension

**Expected behavior**

shown sections and values 

**Screenshots**
see below

**Desktop (please complete the following information):**
 
 - OS: Ubuntu 21.10
 - Gome-Shell: 40.5 
 - Session: ubuntu-xorg
 - Version of lm-sensors (or lm_sensors) 3.6.0

**Additional context**

[image]

HarlemSquirrel, 2021-12-22:
Hi @lemrm could you please share the output of running `sensors` in the terminal?

lemrm, 2021-12-22:
> Hi @lemrm could you please share the output of running `sensors` in the terminal?

... improved the bug report

HarlemSquirrel, 2021-12-22:
Thank you for the quick update! This extension parses fan speed and voltage from `sensors` and I don't see that data in your sensors output. 

You could try running `sudo sensors-detect` which will attempt to find sensors it can read for fan speed and voltage.

If that doesn't work, you may not have the kernel module(s) loaded that you need or such modules may not exist. More information on that can be found on the [ArchLinux Wiki](https://wiki.archlinux.org/title/Lm_sensors)

lemrm, 2021-12-22:
True, only shows voltage, no fan rpm at least.
Is it possible to have something like 

`inxi -Bxxx`
```
Battery:   ID-1: BAT1 charge: 25.8 Wh (97.0%) condition: 26.6/44.7 Wh (59.5%) volts: 8.3 min: 7.6 
           model: SAMSUNG Electronics SR Real Battery type: Li-ion serial: 123456789 status: Unknown cycles: 996
```
for health, cycles, current capacity and voltage?

HarlemSquirrel, 2021-12-22:
Yeah that is something we could add as a new feature to be able to read from there as well. 

I don't have time to add the feature myself but I welcome a pull request! 🙂

lemrm, 2021-12-22:
never coded myself and I'm not familiar with git/collaboration/branches/merge and so on

lemrm, 2021-12-22:
guide me, which file has to be changed

HarlemSquirrel, 2021-12-22:
Here is where we are reading fan and voltage now from sensors output

https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/blob/d422a3db828cabbe9426473a4269caa366748fa8/extension.js#L153-L159

https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/blob/d422a3db828cabbe9426473a4269caa366748fa8/utilities.js#L160-L194

I think we'd want to add new functions to read fan and voltage from `inxi` output in utilities.js and then add that to `fanInfo` and `voltageInfo` in extension.js

A friendly word of caution: writing GNOME JavaScript is not the easiest place to start coding. 
I suggest checking out some basic JavaScript tutorials first at https://www.freecodecamp.org/ and/or https://www.codecademy.com/ and then take a look at the [GNOME guides](https://gjs.guide/guides/). 

I just added a new document on contributing that may help as well 😺